showmewebcam/showmewebcam #149: max resolution

natxopedreira, 2021-07-21:
Hello

I saw here a list of resolutions is possible to set with a HQ camera to max resolution even lowering fps of course, only to take stills for example

[https://github.com/showmewebcam/showmewebcam/blob/1678d4b578c9522068cb5d9312f593a93760f63a/package/piwebcam/multi-gadget.sh#L64](https://github.com/showmewebcam/showmewebcam/blob/1678d4b578c9522068cb5d9312f593a93760f63a/package/piwebcam/multi-gadget.sh#L64)

Thanks!

htruong, 2021-07-21:
Sorry I'm not sure what your question is. Mind to elaborate?

natxopedreira, 2021-07-21:
Sorry, basically whats the max resolution as in mmal can get 4046*3040 and i only saw up to 1080 in that file

dgsiegel, 2021-08-17:
MMAL can in fact get 4046*3040 but only for still photos, for video 1080p is our current maximum.

closed 2021-08-17 by dgsiegel