showmewebcam/showmewebcam #150: Launch Camera-ctl on MacOS for Noobs

j0sc, 2021-07-25:
Hi,

not exactly an issue but as a Terminal Noob I can‘t launch Camera-ctl on MacOS.

Could you please provide a detailed step-by-step guide in the „Customizing camera settings“?

This might sound odd to you but I just don’t get it.

Thanks in advance

dgsiegel, 2021-08-17:
It's described here: https://github.com/showmewebcam/showmewebcam#debugging

What are you struggling with specifically?

j0sc, 2021-08-17:
@dgsiegel 

Hi Daniel,

thanks for your reply.
I really don’t know how to apply these terminal commands.

For example, I‘ve no clue what this means: 

`Use screen, minicom, picocom or the included smwc-expect script to connect to it `
Could you please help by just telling me what commands I need to enter?

Please remember: I‘m a total noob.

Thank you!

jellybaby, 2021-08-17:
Hi,

On a Mac, launch terminal and then use 'ls -l /dev/tty.*' to find the
name of your device:

s -l /dev/tty.*
crw-rw-rw-  1 root  wheel   22,   4 17 Aug 16:57 /dev/tty.usbmodem1424203

Then connect to that device with screen:

$ screen /dev/tty.usbmodem1424203

That should give you a prompt on the Raspberry Pi where you can run camera-ctl

Use Ctrl-A k to exit screen

j0sc, 2021-08-17:
Hi @jellybaby 

thanks a lot!

After I connect with screen, I need to execute 

`/usr/bin/camera-ctl `

in terminal, right?

jellybaby, 2021-08-17:
Hi,

That's right.

Once you have run screen /dev/tty.usbmodemXXXXXX your prompt should be
a # and if you enter the command 'hostname' it should return piwebcam
which confirms you are connected to the Raspberry Pi and not still
your Mac.

j0sc, 2021-08-17:
@jellybaby 

Cool, I see!
Thanks again:)

dgsiegel, 2021-08-17:
@jellybaby thanks for helping out!

closed 2021-08-17 by dgsiegel