showmewebcam/showmewebcam #151: Video can't be initialised on Jetson Nano

greenboards, 2021-07-31:
The camera works great on other Linux machines but I just can't get it to work on a Jetson Nano.

I can see it as video0 but can't seem to initialise it.
I've tried many methods that have been suggested on the NVIDIA form :
https://forums.developer.nvidia.com/t/how-to-find-read-the-contents-of-usb-camera-py-in-the-juperlab/184741

I would really like to use this as a solution to offload some of the image processing for my machine vision project, any suggestions how to get the camera to run?

Thanks

HunterAP23, 2021-11-29:
I think you're using this project incorrectly. The point of this project is provide a minimal OS image that allows a host PC to see the Pi / Jetson Nano as a USB UVC webcam. You really aren't going to be running complex programs like image processing or vision projects - in this case, that stuff would need to be run on the host machine, not the Pi / Jetson Nano.

greenboards, 2021-11-29:
I understand that, and I wasn't trying to attempt that.
I was trying to use it as a webcam only, attached to a Jetson Nano.
The Nano is running inference.

Regards

Murray

HunterAP23, 2021-11-30:
I just want to make sure I understand what you're doing here.
You have the Jetson Nano that has an SD card that's flashed with the latest showmewebcam image, and you have the Pi camera connected to the Jetson Nano with the CSI cable?
Or do you have a Raspberry Pi Zero / Zero w / 4b that has an SD card that's flashed with the latest showmewebcam image, and you have the Pi camera connected to it with the CSI cable, and then the Pi is connected to the Jetson Nano with a USB cable?

greenboards, 2021-11-30:
It's the second, a Pi Zero with CSI camera attached to a Jetson Nano via
USB.