showmewebcam/showmewebcam #153: ADD FEATURE: support monochrom display / eInk display

oOWaXOo, 2021-08-04:
### Some ideas I came up with:

- Show basic status on Display (on, off, connected, used ...)
- Show system information (CPU, RAM, uptime, power)
- Show connection state on display (Network settings in future)
- Show settings on display (brightness, exposure, fps, resolution ...)
- interface to control the displayed text external. (e.g. MQTT)

jim-cathey, 2021-11-11:
It is very easy to talk to e-ink displays via SPI, so this should be very do-able.  Collecting all the state information sounds like the trickiest bit to me.  Formatting of the report screen is straightforward, and there are a number of ways to do it.  One is exemplified in: http://formicapeak.com/~jimc/clunkdraw.html but there are plenty of others.

htruong, 2021-11-12:
Clunkdraw will be very hard to get integrated from my quick look, it's written in Pascal and we don't have a host pascal compiler in buildroot.

Plus we will have to modify the clunkdraw program for it to display anything remotely useful -- this is not hard, just needs someone dedicated enough to do it. 

To be honest, this is low on my list as this is quite niche and requires a lot of effort.

jim-cathey, 2021-11-12:
Yes, not a small job.  The E-Paper (Ink) displays aren't integrated into the existing display system; you have to drive them specially, so no matter what you're going to have at least some additional work.  I keep toying with the idea of porting CD back to C, which is where it started out, but then sanity strikes...

One might have success using the p2c conversion program.  If that worked the output might not be particularly clean, but should be usable.  Especially if one were only cherry-picking a few needed features.

As I see it, the original request requires three tasks/phases:

1) Figure out what stats to gather, and how to do it.  Implement, use printf for testing.
2) Design the report screen(s).  (Sizes, layout, etc.)  What size display panel would be necessary?
3) eInk drawing library.

These needn't all be the same person.  I agree that the cost/benefit ratio is currently not very compelling.