showmewebcam/showmewebcam #158: Uncompressed video formats not showing

MargeritPierre, 2021-09-14:
Hello all, and thank you for this amazing contribution.

I would be interested to stream uncompressed full resolution images from an HQ camera. 
As stated by the README, I created a video_formats.txt in the /boot folder by copying /etc/video_formats.txt, then commented unused formats and added uncompressed options as new lines.

It seems that any uncompressed format is skipped in the file, while changes in mjpeg formats are taken into account.
By browsing the [Add configurable resolutions](https://github.com/showmewebcam/showmewebcam/pull/128) pull request that made possible post-build resolution tuning, I can see that a line may be missing in the multi-gadget.sh file:
`ln -s functions/uvc.usb0/streaming/uncompressed/u functions/uvc.usb0/streaming/header/h` 
Currently only the 'mjpeg' version is included. Would it be the reason for which uncompressed formats cannot be streamed ?

I am using the release image `sdcard-raspberrypi0w-v1.90.img.gz`.

Thank you by advance for your help !

dgsiegel, 2021-09-17:
Yes, that seems to be my fault. Most people don't really use the uncompressed format, so it wasn't tested thoroughly. I've added a PR #161. Would you mind trying it and see if it works for you?

MargeritPierre, 2021-09-20:
Thank you for your reply. Now uncompressed formats are appearing as available ! However, I was only able to make a 640x480 res to work; above the RPi has trouble during initiatialization and does not appear to send frames. 
Would there be something linked to the required frame rates (no frame rate below 15FPS is authorized in multi-gadgets.sh) ? Seems related to the issue [#117](https://github.com/showmewebcam/showmewebcam/issues/117)..
Both RPi0w and RPi4 images (tested on the corresponding boards) are giving the same result, with the HQ camera.

dgsiegel, 2021-09-20:
@htruong unfortunately my PR does not fully close this issue.

@MargeritPierre would you be willing to try what framerates actually do work? We might have to set some framerate limits depending on format and codec. #117 is definitely related and what you're seeing is the hardware limits of your Pi.

MargeritPierre, 2021-09-22:
Ok, so I was trying uncompressed YUV formats with wrong resolutions, as the height has to be a multiple of 16 and the width a multiple of 32 as stated by both the [picamera](https://picamera.readthedocs.io/en/release-1.10/recipes2.html#unencoded-image-capture-yuv-format) and the [RPi camera](https://www.raspberrypi.org/documentation/accessories/camera.html#raspiyuv) documentations.. That is why the 640x480 only was working. Some framerate tests using RPi4 & Rpi0w (no noticeable difference between the two configurations in the results), HQ Camera and the Image Acquisition Toolbox of Matlab, (maybe another software would give different results):
uncompressed 384 288 : 30fps
uncompressed 512 384 : 20fps
uncompressed 640 480 : 13fps
uncompressed 1024 768 : 3.4fps
uncompressed 1280 960 : 3fps
uncompressed 1920 1440 : 1.5fps
uncompressed 2560 1920 : 0.8fps
This seems to indicate that the data rate is around 4MPix/sec. 
Unfortunately higher resolutions than 2560x1920 are not working with my setup (both with RPi0w and RPi4). The RPi manages to initialize, but no image stream is received (and the led indicator stays off).
Thank you again for your help !

dgsiegel, 2021-09-24:
Thanks for testing this.

I think we'll either need to have some way to define fps manuall (e.g. via the video_formats file) or to calculate the fps based on the current resolution automatically.

htruong, 2021-11-09:
V1.91 should have fixed this issue. I'm gonna close this ticket. Please let me know if this doesn't work.

closed 2021-11-09 by htruong