showmewebcam/showmewebcam #168: Support Raspberry Pi Zero 2 W

FLamparski, 2021-11-04:
There's a new Zero out, the processor used is a `RP3A0` (https://www.raspberrypi.com/products/raspberry-pi-zero-2-w/). It boots Raspberry Pi OS Lite. It would be great to add support for this board.

htruong, 2021-11-06:
Working on it.

htruong, 2021-11-08:
I'm perplexed. This change should boot the pi0-2w but it doesn't. If someone has an extra pair of eyes I'd appreciate any suggestion as why this did not boot successfully and stopped at the rainbow screen.

htruong, 2021-11-14:
OK much better, the build I have here is booting the Pi2. Now onto trying to understand why the camera doesn't show up as `/dev/video*`

htruong, 2021-11-14:
Let's see if the builds here work:

https://github.com/showmewebcam/showmewebcam/pull/173

rboerner, 2021-11-16:
I don't know if this is relevant, however just in case it appears that there are issues with the Raspberry Pi Zero 2W and the latest release of Raspberry Pi OS detecting attached cameras.  

There are more details in this thread --> [https://forums.raspberrypi.com/viewtopic.php?t=323462](url)  and in this thread --> [https://forums.raspberrypi.com/viewtopic.php?t=323308](url)

From the second thread:

"We've identified the cause of the camera not being detected on Zero2W - the kernel has been given the wrong GPIO for the camera shutdown line, so it can never bring it out of reset.
An update should be out early next week."

harcomaase, 2021-11-29:
Any news on these activities?

carloe, 2021-11-29:
I finally got my camera working by downgrading the kernel to 5.10.74 as suggested in [this post](https://forums.raspberrypi.com/viewtopic.php?t=323308#p1944586).

`sudo rpi-update ffdfc6e625eb38086b8c15eead2be90acab825c2`

kbingham, 2021-12-02:
To support Bullseye you will need to add libcamera support to the uvc-gadget.

sabotage3d, 2021-12-08:
Is there a test release we can try or we need to compile it from source?

dgsiegel, 2021-12-08:
Each PR builds images, but I doubt they working right now. See #173 / Checks/Artifacts

sabotage3d, 2021-12-09:
Can I make it work manually with this kernel 5.10.74? What are the steps I need to do? I am looking for similar steps to this repo: https://github.com/kinweilee/v4l2-mmal-uvc.

imp1sh, 2022-01-06:
Is there a workaround to make Zero 2 W work?

rscottha, 2022-02-04:
Is Zero 2W support expected soon? I tried to build a pull of the master branch and there is a raspberrypi0_2 directory in the output but no build inside it. Am I building the wrong branch?

dgsiegel, 2022-02-05:
> Is Zero 2W support expected soon? I tried to build a pull of the master branch and there is a raspberrypi0_2 directory in the output but no build inside it. Am I building the wrong branch?

See the linked pull request. There are still a few things to be done, but @htruong is currently busy with personal issues. Happy to review and merge if anybody wants to step up and take over this PR.

Emstr, 2022-04-02:
Any update on support for Zero 2 W? Hard to find the 1 in stock where I live :(

imp1sh, 2024-01-13:
https://github.com/showmewebcam/showmewebcam/issues/220