showmewebcam/showmewebcam #170: Release workflow failed due to sdcard* not found on tagged releases

htruong, 2021-11-09:
https://github.com/showmewebcam/showmewebcam/actions/runs/1437349967/workflow

The sha256sum step failed.

dgsiegel, 2021-11-12:
It's probably, because the wildcard is quoted and hence it's looking for a file with the name `sdcard*` instead of expanding it: https://github.com/showmewebcam/showmewebcam/blob/master/.github/workflows/release.yml#L133

Removing the quotes should make this work.

htruong, 2021-11-12:
That's what I guessed too. I will have to clone to my own repo and test the theory. Blindly change it has broken is once, better be a bit more conservative.

channelbeta, 2021-11-24:
Hi @htruong @dgsiegel , removing the quotes will work.

Evidence: https://github.com/channelbeta/showmewebcam/actions/runs/1230306999/workflow#L133

You can see that this generated a release with checksums correctly.

---
On a side note:

Sorry for the late reply. I was checking the repo every other day to see if/when the action would run and if everything worked correctly. I got busy and stopped checking, then this happened :(

So sorry for the mistake. As you can see on my link above, on my tests I didn't use quotes. Now I can't remember why I added quotes to the filename when raising the PR.

dgsiegel, 2022-02-05:
Added to my newest PR.

closed 2022-02-08 by dgsiegel