showmewebcam/showmewebcam #172: Support for sound output?

jim-cathey, 2021-11-11:
I have been using 1.90 on a Pi 0 1.3 to prototype a barcode-scanner/photo-ID gun, and it's working very well.  Good job on this project, guys, I LOVE the fast boot and the read-only filesystem.  (I'm using uuencode/uudecode to upload binaries to /tmp, which I need for the necessary GPIO work.)  I'm planning for visual feedback via a TFT panel viewfinder, and that appears to be working so far as I can tell, not having the panel actually present yet.  However, I also would like to have audio OUTPUT, for feedback to the operator.  I tried adding "dtparam=audio=on" to the config, but it's looking like maybe the kernel is built without the PWM audio driver at all.

Am I way off base here?  Would it be possible to add enough to the base system to allow for simple .wav files to PWM output?  I'm planning to use one of Adafruit's tiny class-D speaker dongles, and a basic RC PWM filter feeding it.  Clearly I'm not an expert in Raspberry Linux, and all that entails.

Thanks!

htruong, 2021-11-12:
Nice to see you made it that far. A couple of things I have on top of my head:

- If you really just want to hear a simple beep, maybe compiling [a poor man's beep](https://github.com/leon-anavi/rpi-examples/blob/master/buzzer/c/beep.c) and make it a buildroot package is what you need. No need to have hats and stuff.
- I'm not sure what drivers are needed for the HATs to work. Do you know what kernel config is that for the audio PWM? Can you find out what's the driver called? If so, I'm happy to enable that for you.

jim-cathey, 2021-11-12:
Simple beeps are no good.  Current barcode scanners do that, and it's too limiting.  ("Now is be-baw be-baw good?  Or is it beep-beep-beep that's good?"  We're having big problems with that already.)  I want .wav files for our new scanner, those can be much more clear if well-designed.  (A buzzer sound is _never_ good!)

I'm not using any actual HATs, the form factor precludes this.  The peripherals, besides the camera, are all to be hard-wired to the GPIO connector pads.  Currently: trigger switch, TFT display (SPI), analog audio PWM.  There might end up being some LEDs too, not sure yet.

Near as I can tell, the kernel driver is snd_bcm2835, and its configuration parameter is "dtparam=audio=on", which should result in some /dev/snd/XXX devices in the filesystem.  This is not a particularly simple driver, but then neither is v4l2!  My investigation into what aplay needs shows that it wants access to a lot of non-movable configuration files, which is not encouraging with regards to the read-only filesystem.  It may be that I will eventually need a different userland program.  But, none of that matters if the kernel can't/won't drive the Pi's PWM pins!

On the Pi 0 you have to use some alternate pin mappings to get access to the PWM audio signals, but it looks like that already works.  (Consider adding the raspi-gpio program to the filesystem build.  That's how you test/reconfigure the GPIO pins.)  There's just nothing _generating_ PWM signals yet.

Thanks again!

htruong, 2021-11-12:
Let me see what can I do.

By the way for development purposes you can edit the genimage* files and specify the target fs to be ext4 with a specified size. That should allow you to debug easier.

jim-cathey, 2021-11-12:
I'm not actually having much trouble debugging, yet.  I can do most of the work on the Pi 3B+ that's hosting the Pi 0 during development, and then just upload to the Pi 0 for final testing.  (The real host of the Pi 0 will be an Intel NUC.)  With strace (uploaded) and printf I've got all I really need.  The heart of the upload script is:
<PRE>
(echo 'uudecode'
 uuencode $1 /tmp/`basename $1`
 sleep 0.1
 /bin/echo -e \\004
) | sudo tee /dev/ttyACM0 >/dev/null
</PRE>

jim-cathey, 2021-11-14:
The tinyplay program, from the tinyalsa project, seems to be all that is necessary to play .wav files.  (It can't convert formats, but I can do that beforehand on a development system.)  It doesn't need lots of libraries or configuration files, and looks very good for use on this platform.  Now all I need is a functioning sound driver!

jim-cathey, 2022-01-29:
I have been able to use buildroot to work with a modified 1.91 project, and can configure it to include utilities like tinyalsa, strace, etc.  However, no amount of tinkering with configuration (CONFIG_SND_BCM2835) has yet allowed it to include (in the build) the .ko module that actually seems to be responsible for putting the sound driver into the kernel.  I'm mystified, as the exact same flow seems to be what is getting the video driver(s) into the kernel.

jim-cathey, 2022-01-30:
The solution was not entirely obvious, at least to me, but it _is_ possible to include sound in the build. Starting with the 191 image, and buildroot 2021.02.8, there are three major components:

1. Add these two lines to showmewebcam's <CODE>board/linux-base.config</CODE>:
<PRE>
CONFIG_SOUND=y
CONFIG_SND=y
</PRE>
Both are necessary in order to get the kernel's sound driver included in the build.

2. Add (at least) these two lines to showmewebcam's <CODE>configs/config</CODE>:
<PRE>
BR2_PACKAGE_ALSA_LIB=y
BR2_PACKAGE_TINYALSA=y
</PRE>
This adds tinyplay to the build, a simple CLI utility for playing sound files. (You may wish to add more utilities—I did.)

3. Add this to <CODE>/boot/config.txt</CODE> on the Pi 0:
<PRE>dtparam=audio=on
[pi0]
# Enable PWM audio on alternate pins
dtoverlay=audremap,enable_jack=on,pins_18_19
[all]
</PRE>
This enables the audio hardware, and on the Pi 0's drives its alternative PWM sound pins. (The normal pins aren't connected to anything on Pi 0.)
<P>
With all these done and an audio amplifier connected, and a way to load sound files on the target hardware, you can then play sounds on the webcam.