showmewebcam/showmewebcam #175: getting error while building image file as mentioned in Development & building section

Parikshit01, 2021-12-07:
While following the mentioned steps in the repo's Development & building section, After running` ./build-showmewebcam.sh raspberrypi0w` command i'm getting below error.
```
ERROR: file(rpi-firmware/bootcode.bin): stat(/home/parikshit/showmewebcam/output/raspberrypi0w/images/rpi-firmware/bootcode.bin) failed: No such file or directory
ERROR: vfat(boot.vfat): could not setup rpi-firmware/bootcode.bin
```

kindly let me know possible reason and solution for this.

Parikshit01, 2021-12-07:
output of `./build-showmewebcam.sh raspberrypi0` is as below
`grep: /home/parikshit/showmewebcam/output/raspberrypi0/images/rpi-firmware/config.txt: No such file or directory
INFO: cmd: "mkdir -p "/home/parikshit/showmewebcam/output/raspberrypi0/build/genimage.tmp"" (stderr):
INFO: cmd: "rm -rf "/home/parikshit/showmewebcam/output/raspberrypi0/build/genimage.tmp"/*" (stderr):
INFO: cmd: "mkdir -p "/home/parikshit/showmewebcam/output/raspberrypi0/build/genimage.tmp"" (stderr):
INFO: cmd: "cp -a "/tmp/tmp.WXdBngpahk" "/home/parikshit/showmewebcam/output/raspberrypi0/build/genimage.tmp/root"" (stderr):
INFO: cmd: "find '/home/parikshit/showmewebcam/output/raspberrypi0/build/genimage.tmp/root' -depth -type d -printf '%P\0' | xargs -0 -I {} touch -r '/tmp/tmp.WXdBngpahk/{}' '/home/parikshit/showmewebcam/output/raspberrypi0/build/genimage.tmp/root/{}'" (stderr):
ERROR: file(rpi-firmware/bootcode.bin): stat(/home/parikshit/showmewebcam/output/raspberrypi0/images/rpi-firmware/bootcode.bin) failed: No such file or directory
ERROR: vfat(boot.vfat): could not setup rpi-firmware/bootcode.bin
make[1]: *** [Makefile:823: target-post-image] Error 1
make: *** [Makefile:23: _all] Error 2
make: Leaving directory '/home/parikshit/showmewebcam/output/raspberrypi0'

`

dgsiegel, 2021-12-08:
Which buildroot version did you use? Wanna try with an older stable, 2021.02.4? 

Buildroot is currently restructuring a few things related to the Raspberry and we haven't catched up yet, waiting for #173

Parikshit01, 2021-12-08:
thanks. solved with the latest stable repo.
I was cloning buildroot latest Github repo which was probably not a stable release.

closed 2021-12-08 by Parikshit01