showmewebcam/showmewebcam #178: High CPU usage after streaming

Janghou, 2021-12-10:
CPU usage Zero (W):

- After boot (idle): < 5%
- Streaming: ~ 70%
- After stream has stopped (idle again): ~ 55%

Reported by top for `/opt/uvc-webcam/uvc-gadget`

Tested on Ubuntu (21.10)

okawo80085, 2021-12-12:
Would be nice to see some sort of work around/solution to this, in my case the rpi stays plugged in indefinitely, i don't want to waste more power when the device is idle, i use it in a VR system, wasting power is the last thing i need. If its possible to enforce that state via serial commands that works for me too.

dgsiegel, 2021-12-12:
You might want to report this to https://github.com/peterbay/uvc-gadget

okawo80085, 2021-12-12:
> You might want to report this to https://github.com/peterbay/uvc-gadget

Will do just that, thanks :)

closed 2022-02-05 by dgsiegel