showmewebcam/showmewebcam #179: Does this support v4l2loopback + usb gadget?

wuhanstudio, 2021-12-11:
Hi,

I'm wondering does this support streaming video from **v4l2loopback dummy devices**, so that we can add some preprocessing or **filters**, for example:

[image]

https://github.com/likeablob/rpi-wfh-webcam

I find this project, but it only works for the CSI camera. I can olny see a **black screen** on Windows while redirecting from a **USB camera** or **v4l2loopback devices** on raspi4.

[image]

Is it possible to redirect from USB camera (/dev/video3) to v4l2loopback (/dev/video0) after applying filters, and then to USB gadget (/dev/video2)?

```
$ v4l2-ctl --list-device

fe980000.usb (gadget):
        /dev/video2

mmal service 16.1 (platform:bcm2835-v4l2-0):
        /dev/video1

Dummy video device (0x0000) (platform:v4l2loopback-000):
        /dev/video0

USB2.0 PC CAMERA: USB2.0 PC CAM (usb-0000:01:00.0-1.4):
        /dev/video3
        /dev/video4
```

notsig11, 2022-01-07:
Have you had any luck with this?  I was hoping to do something similar with my rpi4...

wuhanstudio, 2022-01-07:
Here are my test results:

[image]

It looks like the uvc-gadget redirects raw video data withou encoding. Since v4l2loopback virtual devices only support YUYV format, maybe one solution is to encode the raw data using MJPEG before redirecting.

wuhanstudio, 2022-01-29:
Hi, @notsig11 Finally it worked !!

The left side is the original image from a USB camera (YUYV) captured by the pi4, and the right side is the output of the simulated camera (MJPEG) on Windows.

https://user-images.githubusercontent.com/15157070/151670268-a14e5987-a173-4980-a80c-3410dcd67db4.mp4

Sorry that the code is still quite messy.

https://github.com/wuhanstudio/adversarial-camera