showmewebcam/showmewebcam #185: uvc-gadget exits upon starting a high res capture even when enough gpu_mem is available

wherget, 2022-01-16:
(This is a report [submitted by [vogarth] on Discord](https://discord.com/channels/774949618832113674/774953831267958795/931823706131529729))

## Expected Behavior

(This assumes an HQ sensor which can deliver said resolution, Ed.)
I would expect asking for uncompressed 4096x3040 (black borders where sensor data doesn't exist is just fine, I'd rather see this than upscaled images which is what raspistill returns), to return frames of that size, at whatever speeds the bandwidth allows.

## Current Behavior

(with gpu_mem set to 192 on a Pi4, Ed.) Instead of crashing, uvc-gadget sets the camera up properly, sets the buffers up properly, then exits gracefully, killing any further camera and serial interaction until a power cycle.

## Steps to Reproduce
1. HQ camera on Pi4
2. gpu_mem to 192
3. add uncompressed resolution to /boot/video_formats.txt
4. start capture session

## Editors remarks

Related to #184, even with supposedly enough memory present, it still doesn't work. I'm filing it here as well, since it's not yet clear whether it's a purely `uvc-gadget` issue or whether the kernel/mmal layer might be involved as well.

I'm also a bit stumped by the "serial cutting out" which still hints at something in the kernel (usb stack?) going haywire. I don't think USB serial should be impacted by just uvc-gadget dying unexpectedly. [vogarth] retrieved the logs (which is how they got the "sets up buffers properly") over wifi IIRC.

wherget, 2022-01-17:
possibly related: https://github.com/peterbay/uvc-gadget/issues/19 (which is the intent of the expected bahaviour, if maybe not the actual issue we're seeing here)

atalkingegg, 2022-01-21:
A few notes: please correct me if I'm wrong, but uvc-gadget operates the USB-serial host as well as the UVC data and control pipelines, so when uvc-gadget crashes or exits, the client can no longer see video or connect to a console (via USB-serial). If wifi had previously been configured, it's still possible to get a console and look at dmesg / restart uvc-gadget manually that other channel.

Raspistill (which also uses video4linux frameworks) on a Raspberry Pi 4 2GB - 'Buster' - will NOT capture 4032x3040 images unless gpu-mem is set to 192, then it works fine. On a Raspberry Pi Zero W with latest stock ShowMeWebCam, uvc-gadget causes a kernel memory oops and crashes if it tries to switch into that or other higher resolution modes, that's with gpu_mem at 100. Set to 192 it appears to successfully change camera resolutions, successfully sets up some buffers, then it gracefully quits.
This strongly suggests this is strictly a uvc-gadget "decision" to exit, not an underlying Linux / v4l2  framework issue. Should be fixable!
Best of luck and thanks!

wherget, 2022-01-23:
As I've stated above, uvc-gadget should have nothing to do with USB serial. The device is configured separately in [multi-gadget.sh](https://github.com/showmewebcam/showmewebcam/blob/master/package/piwebcam/multi-gadget.sh), and then handled by a plain old getty process, presumably originating from the buildroot getty config in the [config file](https://github.com/showmewebcam/showmewebcam/blob/master/configs/config#L9). If you `grep` the uvc-gadget source you will find no mention of "tty" anywhere, and the only mention of "serial" is the gadget device binding you see above in mult-gadget.sh, too.

While I don't doubt that you can't connect to USB serial anymore once whatever causes uvc-gadget to exit happens, looks more like a correlation instead of a causal relationship to me.

kbingham, 2022-01-23:
I'm working on a theory and memory from when I've worked on this in the past, but when UVC gadget exits, it likely blocks the entire USB UDC.
To resume serial, without a UVC I think you'd have to reconfigure the UDC to remove the (non functioning) UVC ... or update the kernel drivers so they allow a disconnected gadget to not block the other gadgets. UVC is rather special in the way that it can block waiting on userspace interactions. Perhaps there might be ways to add timeouts or some thing else to prevent the UVC having such an impact on the overall state of the device, but I haven't investigated it.

But to clarify above, UVC-gadget does not operate the serial, it's just that it shares the same UDC. If you lock up the drivers waiting on a (non present) uvc-gadget application, then it can block other gadgets. I'd consider that something that would be great to fix, but requires kernel development and investigation to determine if it's possible to improve.