showmewebcam/showmewebcam #186: bringing up the camera in a repeatable state (esp. concerning white balance and gain)

wherget, 2022-01-16:
(this is another one [submitted by [vogarth] on Discord](https://discord.com/channels/774949618832113674/774953831267958795/931822495869325322))

## Expected Behavior

I would expect that a camera in manual mode with the same settings to behave exactly the same every time no matter what lighting environment it was started up with.

## Current Behavior

if you set a camera in full manual mode, then bring up the camera in darkness, it will have nearly quadruple the light sensitivity as the same camera started up under bright light, both times with the same exact settings.

I [posted some histogram images](https://discord.com/channels/774949618832113674/774953831267958795/921474372173180928) in this forum on Dec 17 showing this, and how repeatable it is.

camera started in darkness, light turned to "normal", target is white paper with a few black lines, camera in full manual mode, white balance, brightness and contrast dialed in:
[image]

camera reset in high brightness environment, then lighting brought back to exactly where 1st image came from
[image]

This behavior shows up with raspistill, so is likely to be in the ~~v4l2~~ mmal ([raspistill](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c) doesn't use v4l, Ed.) code. 

## Editors remarks

Since this also happens with pure mmal raspistill, I'm at a bit of a loss on how we could fix this. Maybe there's code paths to reset the sensor's gain controls we could explore?

kbingham, 2022-01-17:
This should be solved by the newer libcamera pipelines on Raspberry Pi as we always aim to start with a consistent camera state. Unfortunately that doesn't help you directly yet as there isn't yet uvc-gadget support for libcamera.

wherget, 2022-01-17:
I agree, the much more low-level control the unicam/libcamera stack enables (but maybe also requires?) sounds like the most likely venue. I've taken the liberty of submitting a corresponding issue.

FilipDominec, 2025-05-22:
@wherget , how did you get ```raspistill``` working in the showmewebcam system? It seems not to be present in the current version. Did you recompile the entire embedded system to include it? I cannot see such a change in your fork, yet it would be very useful to try it myself.