showmewebcam/showmewebcam #191: Correcting lens distortion - Possible?

mlinton, 2022-02-24:
First, big thanks for the release of this software.  I'm using this solution as a USB web cam and the 6mm lens I'm using has some barrel distortion that I'd like to get rid of.  Is there a suggested way to do this?  Some method of calibrating the camera and lens using software?

I did find a link to the OpenCV Calibration - but not sure how this would apply in this case.  https://github.com/jeeho-ahn/OpenCV_Camera_Calibration_Undistort_Video

Any tips would be helpful.

dgsiegel, 2022-02-25:
Unfortunately right now there is no way to correct it. Most people just switch to a lens with less barrel distortion.

But there were some ideas to add filters via v4l2loopback, like https://github.com/showmewebcam/showmewebcam/issues/179

mako220, 2022-04-16:
For now, you can use OBS to correct the distortion on your pc and then use OBS's virtual camera functionality to use the corrected image. OBS is a free software, available on all platforms (windows, linux and mac os). To set a custom filter in OBS, you need streamfx plugin. Watch [this](https://www.youtube.com/watch?v=Y8RxMtzr23o) video to how to proceed.