showmewebcam/showmewebcam #196: start.elf: is not compatible

torniker, 2022-04-24:
```
start.elf is not compatible 
this board requires newer software
```
[image]

This is the message I get after installing 1.91 version on the sdcard.
the device is Pi 4B. In case any more information is needed please let me know.

Martin-HiPi, 2022-05-06:
General note about current stocks of Raspberry Pi 4s:
While the global Raspberry Pi supply is still extremely limited, it's important to note that the Pi4 boards being manufactured now are hardware revision 1.5. This hardware revision requires updated OS boot files, and this affects customized images like the one you use.

Here is a link that describes the two file sets in the card image that must be updated: https://forums.raspberrypi.com/viewtopic.php?f=28&t=315742#p1890907. The post is from July 19, 2021 and lists start*.elf and fixup*.dat as the two sets to update.

Unexpected hardware changes like this are disruptive. To be warned in advance, we recommend signing up for Product Change Notifications (PCN's) in the Raspberry Pi Foundation's Product Information Portal. To do this, register a free account at the 'restricted' level, and then enable notifications for each individual product that you use. https://pip.raspberrypi.com/restricted

dgsiegel, 2022-05-06:
I am currently trying to rebase the current Buildroot/RPi Firmware here #197. Unfortunately there were several changes in the V4L2 firmware/kernel module as well, which make the transitions a little bit harder.

kbingham, 2022-05-09:
If you need support getting this converted to libcamera let me know. I've long followed this project, and I don't want to see it fall behind by not being able to update to the new interfaces, so I'll help where I can and offer advice if requested.

dgsiegel, 2022-05-09:
> @dgsiegel If you need support getting this converted to libcamera let me know. I've long followed this project, and I don't want to see it fall behind by not being able to update to the new interfaces, so I'll help where I can and offer advice if requested.

@kbingham sure, that would be great! I've managed to build a bootable image, unfortunately I have troubles loading up both camera devices (the actual camera, as well as the UVC device through ConfigFS).

Using the legacy mode, the camera gets just detected fine, but ConfigFS is unable to create a device which can be consumed by uvc-gadget.

Using libcamera, the firmware gets loaded and I can see both /dev/video0 and /dev/video1, but both can't be used.

Shall we move the discussion to #197? You can also ping me on the discord channel. Thanks!!

kbingham, 2022-05-09:
> Using libcamera, the firmware gets loaded and I can see both /dev/video0 and /dev/video1, but both can't be used.
Please be aware you can not simply use /dev/videoX and the v4l2 calls directly with the libcamera stack.

The legacy stack includes *a lot* of extra processing that occured on the GPU in proprietary code. That's now all moved to the linux stack, and is run by libcamera - but that means using calls to libcamera - *not* v4l2 to control the camera.

There 'is' a v4l2 adaptation layer, but I don't think that's going to be approprate long term here. It might be possible to attempt using it though - "libcamerify uvc-gadget" would enable it. 

(as in - there should be a script called libcamerify installed, which you can use to run uvc-gadget - and that will expose a v4l2-interface to the camera)

AudiThisWorld, 2022-09-26:
I have several Pi Zero 2 W's, but after many hours, I couldn't get it to work. I couldn't get my hands on a Pi Zero v1.3 or Pi Zero W, so I bought a Pi 4+, but it also doesn't work due to this change. I swear, I have the worse luck, and I don't know what to do now. =(