showmewebcam/showmewebcam #200: not showing up on ubuntu

ejooPasco, 2022-05-20:
I build a webcam with sdcard-raspberrypi0w-8e9f9ac.img.zip on raspberry pi zero w, not 2. It shows up on Win 10, no preview, but after selecting it in e.g. Win camera app it works just fine. 

output of sudo demsg -w: 

[40707.495196] usb 3-3: new full-speed USB device number 2 using xhci_hcd
[40712.427209] usb 3-3: new high-speed USB device number 3 using xhci_hcd
[40712.576109] usb 3-3: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
[40712.576118] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[40712.576121] usb 3-3: Product: Piwebcam
[40712.576124] usb 3-3: Manufacturer: Show-me Webcam Project
[40712.576126] usb 3-3: SerialNumber: 000000003f83db4b
[40712.609108] mc: Linux media interface: v0.10
[40712.609466] cdc_acm 3-3:1.2: ttyACM0: USB ACM device
[40712.609495] usbcore: registered new interface driver cdc_acm
[40712.609497] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[40712.618146] videodev: Linux video capture interface: v2.00
[40712.637051] usb 3-3: Found UVC 1.00 device Piwebcam (1d6b:0104)
[40712.640784] usb 3-3: Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).
[40712.645343] usb 3-3: Failed to query (GET_INFO) UVC control 9 on unit 2: -32 (exp. 1).
[40712.647616] usb 3-3: Failed to query (GET_INFO) UVC control 10 on unit 2: -32 (exp. 1).
[40712.656228] usb 3-3: Failed to query (GET_INFO) UVC control 4 on unit 2: -32 (exp. 1).
[40712.663734] input: Piwebcam: UVC Camera as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/input/input39
[40712.663862] usbcore: registered new interface driver uvcvideo
[40724.660945] usb 3-3: Failed to query (GET_DEF) UVC control 6 on unit 2: -32 (exp. 2).

Do others have the same problem?

What's to do next?

ejooPasco, 2022-05-21:
Forgot to mention: raspi HQ camera is mounted. I think may be I should disable some controls. But I have no idea how the above numbers 6, 9, 10 and 4 relate to the bit sums on https://github.com/showmewebcam/showmewebcam/wiki/UVC-Controls. 
Is this the right direction for troubleshooting?

ejooPasco, 2022-05-23:
As shown in the page UVC-Controls linked above I added 
usb_f_uvc.camera_terminal_controls=2,0,0 usb_f_uvc.processing_unit_controls=11,0
to the cmdline.txt file. This exposes uvc controls for brightness, contrast and saturation to the host. Other settings did not work for me. Now its booting fine!

ejooPasco, 2022-05-23:
Hmm this was wrong here. sorry. Is there such thing as a forum?

dgsiegel, 2022-05-23:
Yeah, in most cases it's the cable. As you discovered the UVC controls only control what settings will be exposed. Feel free to join our discord channel.

closed 2022-05-23 by dgsiegel