showmewebcam/showmewebcam #201: Install Arducam drivers as well?

RoganDawes, 2022-06-14:
Hi folks,

I have purchased a couple of Arducam cameras in their recent kickstarter. These are super high resolution, and super tiny, etc. However, they need special drivers, as per this page: https://www.arducam.com/docs/cameras-for-raspberry-pi/raspberry-pi-libcamera-guide/

And yes, while this needs libcamera, and I know that uvc-gadget does not yet support libcamera, I see that after installing the above drivers, I now get a `/dev/video0` that I never did before, using the available downloads from this project. So, I guess I am asking whether it may be possible to get a build such as `sdcard-raspberrypi0-arducam-8e9f9ac.img.zip`, that has the Arducam drivers preloaded instead of the usual Pi Foundation-provided ones?

I will check to see whether it is just the kernel drivers that are required to get `/dev/video0` to show up, or whether the entire Glamor/KMS steps are required.

RoganDawes, 2022-09-14:
Possibly relevant? https://www.cnx-software.com/2022/09/13/picamera2-python-camera-library-for-raspberry-pi-leverages-libcamera-open-source-framework/

kbingham, 2022-09-14:
Has anyone tried to 'libcamerify uvc-gadget' ?

I made the libcamerify script to help test the v4l2 adaptation layer, which should work with uvc-gadget' I expect. It won't support changing controls (yet) but it might be an interim solution.

We're hoping to get an updated release of our uvc gadget with libcamera support out in the nearish future, but I'm not sure how long it will be untill we can start working on it.

dgsiegel, 2022-09-14:
> Has anyone tried to 'libcamerify uvc-gadget' ?

Yeah, when we've been talking on discord. Unfortunately I hadn't much success and kept testing with the legacy drivers.