showmewebcam/showmewebcam #205: SSL certificate problem during buildroot

codepoet80, 2022-08-29:
Executing the command `git clone --recurse-submodules https://github.com/showmewebcam/showmewebcam.git` almost immediately resulted in: `fatal: unable to access 'https://git.buildroot.net/buildroot/': SSL certificate problem: certificate has expired`

I was able to work-around by removing the folder, then retrying with: `GIT_SSL_NO_VERIFY=true git clone --recurse-submodules https://github.com/showmewebcam/showmewebcam.git`
Not sure how safe this is, though!

GDGooD, 2024-11-09:
It happened again today - git.buildroot.net certificate has expired yesterday and can not clone submodule.

GDGooD, 2024-11-09:
Shall be fixed by this pull request: https://github.com/showmewebcam/showmewebcam/pull/241