showmewebcam/showmewebcam #210: Unable to use RPi zero v1.3 with arducam IMX219 module in ubuntu

Varghesemela, 2022-10-17:
Hi,
I tried installing the latest release on showmewebcam on RPi zero v1.3 with the IMX219 camera module. When i connect the USB to my ubuntu computer, the led blinks once then 3 times to confirm proper booting of the RPi with the camera.
But regardless of that, i am unable to get any feed from the camera, i can even communcate with the debug line on ttyACM0, but i am not able to see any image or video.
I tried using `mpv /dev/video0`
but after waiting for 30 secs, i get a response 
> Playing: /dev/video1
> [ffmpeg/demuxer] video4linux2,v4l2: ioctl(VIDIOC_STREAMON): Connection timed out
> [lavf] avformat_open_input() failed
> Failed to recognize file format.
it doesnt work on cheese or guvcview also

the camera module is the one below:
[https://www.arducam.com/product/arducam-raspberry-pi-camera-v2-8mp-ixm219-b0103/](url)

misel228, 2022-10-17:
Showmewebcam still uses the (somewhat proprietary) tools directly from the Pi Foundation that are only available for the official Pi cameras.

ArduCams use the Video4Linux framework which is the de facto standard for Cameras on Linux.

IIRC there is a V4L version in the works but I don't know its current status.

kbingham, 2022-10-17:
From : https://www.arducam.com/product/arducam-raspberry-pi-camera-v2-8mp-ixm219-b0103/

Note: This camera module supports the latest Raspberry Pi OS Bullseye (released on Jan 28th, 2022) and libcamera apps, for the previous Raspberry Pi OS (Legacy) users, contact us at sales@arducam.com.

Most arducam cameras require libcamera, as they're not supported by Raspberry Pi.
showmewebcam doesn't yet have libcamera support.

(Though we've made a lot of progress on our uvc-gadget, which I expect to then add libcamera support to in the nearish future).

dgsiegel, 2022-10-17:
> (Though we've made a lot of progress on our uvc-gadget, which I expect to then add libcamera support to in the nearish future).

Let us know once that is ready, would love to integrate it in showmewebcam.

Varghesemela, 2022-10-17:
OK thanks! Eagerly waiting for libcamera support on showmewebcam. Until then, i will have to work with the official rpi camera v2.