showmewebcam/showmewebcam #211: Pi0 not recognized as camera anymore

xammmue, 2022-10-17:
Hello,

I setup a pi0 with showmewebcam some days ago and mounted it to my wall. It worked flawlessly for some time, but yesterday suddenly stopped being recognized. I had it connected to a Pi4 running hyperHDR.

I since tried connecting it to my computer, flashing the microSD card again and also testing the microSD card as well es the camera with another pi0W. I also tested the other camera which was connected to the pi0W with my pi0. It seems I just can't get the pi0 to be recognized by any pc anymore, while the camera as well as the microSD work just fine with my pi0W.

Based on that I would assume a device defect, but I just received that pi0 last week and didn't even though it while it stopped working. Booting it up also seems to work fine, as the led still lights up as usual, even flashing 3 times in the end. The indicator led of the camera lights up 2 times during booting up but does not stay on, probably because it isn't used.

Do you have any idea what could be the issue here?

PSF1, 2024-09-14:
Hi, I have the some problem after 3 or 4 years working.

I flash a the new version but it didn't work. The command `$ sudo dmesg -w` show:
```
[  265.981803] usb 5-1: new full-speed USB device number 9 using xhci_hcd
[  268.861735] usb 5-1: new high-speed USB device number 10 using xhci_hcd
[  269.017059] usb 5-1: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
[  269.017065] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  269.017069] usb 5-1: Product: Piwebcam
[  269.017073] usb 5-1: Manufacturer: Show-me Webcam Project
[  269.017075] usb 5-1: SerialNumber: 00000000332c3aa7
[  269.060139] cdc_acm 5-1:1.0: ttyACM0: USB ACM device
```
But don't create the video devices.

PSF1, 2024-09-15:
I changed the camera, and camera wire, by a new one and the now it's working.

```
[ 1348.572248] usb 5-1: new full-speed USB device number 19 using xhci_hcd
[ 1353.236000] usb 5-1: new high-speed USB device number 20 using xhci_hcd
[ 1353.387568] usb 5-1: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
[ 1353.387576] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1353.387580] usb 5-1: Product: Piwebcam
[ 1353.387583] usb 5-1: Manufacturer: Show-me Webcam Project
[ 1353.387586] usb 5-1: SerialNumber: 00000000332c3aa7
[ 1353.482664] cdc_acm 5-1:1.2: ttyACM0: USB ACM device
[ 1353.512264] videodev: Linux video capture interface: v2.00
[ 1353.523235] usb 5-1: Found UVC 1.00 device Piwebcam (1d6b:0104)
[ 1353.526771] usb 5-1: Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).
[ 1353.533398] usb 5-1: Failed to query (GET_INFO) UVC control 9 on unit 2: -32 (exp. 1).
[ 1353.535772] usb 5-1: Failed to query (GET_INFO) UVC control 10 on unit 2: -32 (exp. 1).
[ 1353.543150] usb 5-1: Failed to query (GET_INFO) UVC control 4 on unit 2: -32 (exp. 1).
[ 1353.551992] input: Piwebcam: UVC Camera as /devices/pci0000:00/0000:00:08.1/0000:2d:00.3/usb5/5-1/5-1:1.0/input/input18
[ 1353.552119] usbcore: registered new interface driver uvcvideo
```