showmewebcam/showmewebcam #213: Rpi Zero W Creating 'Failed to query (GET_INFO) UVC control' Error on Rpi 3 Buster 3 and Rpi 4 Bullseye 64 bit

embgit, 2022-10-18:
Rpi Zero W USB camera works on Windows PC and Surface with no problems.

Ran 'sudo dmesg -w' on Rpi 3 Buster and Rpi4 Bullseye 64 bit

[86890.861549] rpi_firmware_get_throttled: 7 callbacks suppressed
[86890.861556] Under-voltage detected! (0x00050005)
[86897.101515] rpi_firmware_get_throttled: 7 callbacks suppressed
[86897.101523] Voltage normalised (0x00000000)
[86897.541429] usb 1-1.2: new full-speed USB device number 8 using dwc_otg
[86902.411452] usb 1-1.2: new high-speed USB device number 9 using dwc_otg
[86902.542438] usb 1-1.2: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
[86902.542456] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[86902.542465] usb 1-1.2: Product: Piwebcam
[86902.542474] usb 1-1.2: Manufacturer: Show-me Webcam Project
[86902.542484] usb 1-1.2: SerialNumber: 000000007197e5d8
[86902.547643] uvcvideo: Found UVC 1.00 device Piwebcam (1d6b:0104)
[86902.555457] uvcvideo: Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).
[86902.564847] uvcvideo: Failed to query (GET_INFO) UVC control 9 on unit 2: -32 (exp. 1).
[86902.567369] uvcvideo: Failed to query (GET_INFO) UVC control 10 on unit 2: -32 (exp. 1).
[86902.574335] uvcvideo: Failed to query (GET_INFO) UVC control 4 on unit 2: -32 (exp. 1).
[86902.584688] uvcvideo 1-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[86902.584706] uvcvideo 1-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[86902.585096] input: Piwebcam: UVC Camera as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/input/input3
[86902.590795] cdc_acm 1-1.2:1.2: ttyACM0: USB ACM device

embgit, 2022-10-19:
Forgot to mention. Using Rpi HQ camera, Not Arducam.

mat-l, 2023-03-09:
Same for me on Ubuntu 20.04.5 LTS

[ 1101.682132] usb 1-6: USB disconnect, device number 17
[ 1112.178293] usb 1-6: new full-speed USB device number 18 using xhci_hcd
[ 1117.161922] usb 1-6: new high-speed USB device number 19 using xhci_hcd
[ 1117.315603] usb 1-6: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
[ 1117.315618] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1117.315624] usb 1-6: Product: Piwebcam
[ 1117.315628] usb 1-6: Manufacturer: Show-me Webcam Project
[ 1117.315632] usb 1-6: SerialNumber: 0000000006d9a99f
[ 1117.319046] usb 1-6: Found UVC 1.00 device Piwebcam (1d6b:0104)
[ 1117.338633] usb 1-6: Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).
[ 1117.344769] usb 1-6: Failed to query (GET_INFO) UVC control 9 on unit 2: -32 (exp. 1).
[ 1117.347259] usb 1-6: Failed to query (GET_INFO) UVC control 10 on unit 2: -32 (exp. 1).
[ 1117.350950] usb 1-6: Failed to query (GET_INFO) UVC control 4 on unit 2: -32 (exp. 1).
[ 1117.364094] input: Piwebcam: UVC Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/input/input29
[ 1117.365457] cdc_acm 1-6:1.2: ttyACM0: USB ACM device