showmewebcam/showmewebcam #214: Support other raspberry pi

hedjour, 2022-10-20:
Is it planified to support older Raspberry pi like pi 2 b or the possibility to compile a set for these Pi ?

misel228, 2022-10-20:
It's technically impossible. The main chip on all Raspberry Pis until the 3 has only one USB port than can run either as host or device (the latter is needed for showmewebcam).

For all boards of the B type this port is connected to another chip which drives Ethernet and the additional USB ports. There it isn't possible to switch.

On the A boards it would be theoretically possible to switch the single USB port into device mode but due to an error/oversight in the board design it's not possible for the 3A(+) models. I don't know about the 1A models and don't think there are 2A(+) models at all.

With the Pi4 the situation is different. The Ethernet and USB ports are connected to the chip via PCIe and not USB. The USB port of the chip is connected to the USB-C charging port that can be switched into device mode.

*Disclaimer* This information is from the top of my head and it maybe incomplete or incorrect in some parts.

hedjour, 2022-10-20:
Many thanks for these explanations.

closed 2022-10-20 by hedjour