showmewebcam/showmewebcam #217: Support for Pi Camera 3

dansinclair25, 2023-01-12:
I recently purchased the new Pi Camera 3, and after reading the docs it looks like it doesn't use the same libs as the previous versions. Is there a plan to support the camera 3 at all?

kbingham, 2023-01-12:
"yes".

joriswvanrijn, 2023-04-05:
Would love that as well!

kbingham, 2023-04-06:
https://gitlab.freedesktop.org/camera/uvc-gadget works with libcamera and the Raspberry Pi cameras v3

willnotremember, 2023-05-31:
Hello there! 
I hope it's ok to ask some advice, because I have been following this tutorial: https://www.raspberrypi.com/tutorials/plug-and-play-raspberry-pi-usb-webcam/ that uses the uvcgadget (@kbingham ) but unfortunately on my Raspberry Pi 4 with a Raspberry Pi Camera V3 Wide Noir it is not working. In VLC it shows "UVC Camera" but won't open a stream. Is there anything I have to be aware of? 
Thank you so much!

> https://gitlab.freedesktop.org/camera/uvc-gadget works with libcamera and the Raspberry Pi cameras v3
> 
> This needs integration with SMWC perhaps in a branch and then further development to consider other features that SMWC had which may likely not be supported in this version of uvc gadget. Uvc-gadget is now a library and the fire code can be used by another/ custom application to implement other features for SMWC

kbingham, 2023-05-31:
@willnotremember Are you using Show me webcam? or just uvc-gadget directly? If you're just using uvc-gadget directly and having issues - then please raise an issue at https://gitlab.freedesktop.org/camera/uvc-gadget/-/issues/new

willnotremember, 2023-05-31:
Thank you for the fast reply! I actually tried both  –  with the same outcome. Other cameras worked, so it's most likely the Module 3 :-/ Will raise the issue!

kbingham, 2023-05-31:
Please make sure you provide full logs and output of what messages you get from the uvc-gadget application.

willnotremember, 2023-05-31:
Hhm that is a bit the point, everything seems to be working and I receive no error messages. There is just no stream after plugging the Raspberry Pi with V3 in.

kbingham, 2023-07-11:
Yes, unfortunately - the hardware JPEG encoder is not exposed in the new camera stack from Raspberry Pi - so it's falling back to software encode and isn't as good performance on the Pi Zero 1st gen. I've used it on the Pi Zero 2 W and it's usable with the v3.

Interestingly though - I thought the hardware encoder wasn't exposed any more - but it looks like there might be something at /dev/video31 - bcm2835-codec-encode_image which has a JPEG output - so we might be able to get that integrated!

acketon, 2023-07-11:
By usable with a Pi Zero 2 W, how would you describe the performance? Any lag? Is it worth buying an overpriced Pi Zero 2 on ebay?

I have a PI Zero W 1 w/ Camera Module 2 built into an Apple iSight case that I've been using as my webcam (based on https://github.com/maxbbraun/pisight) but I've been interested in upgrading it to this Camera Module 3 and seeing if I could get a better quality image out of it.

kbingham, 2023-07-11:
I would never suggest buying an overpriced Pi on ebay - the real retailers are getting stock lately - so if you want to try it - get one from an official source. https://rpilocator.com/?cat=PIZERO2 is worth checking from time to time.

The v3 camera is definitely better quality than the v2 - but on the Pi0-2W it was still a bit choppy with not quite 30FPS - but it was very usable for video calls last I used it. If you want to see the quality before buying one to play with - message me privately and I'll join a video call to you with the device.

There's definitely more that could be done to improve it if someone is willing to spend time looking into it.