showmewebcam/showmewebcam #220: Project dead?!

imp1sh, 2023-08-09:
To me this project seems dead. Is it?

imp1sh, 2023-08-09:
For starters the documentation page is down and the issue for the Pi Zero 2 is open for nearly two years now.

kbingham, 2023-08-09:
Do you have a pi zero 2 ? Are you willing to work on it?

kbingham, 2023-08-09:
Which documentation page is down ? - Isn't it listed at https://github.com/showmewebcam/showmewebcam/wiki ?

kbingham, 2023-08-09:
https://github.com/showmewebcam/showmewebcam/pull/197#issuecomment-1632062869 Pi Zero 2 support is here - but the performance isn't quite up to scratch yet. It needs someone to look at integrating the hardware encoders to make it usable.

imp1sh, 2023-08-09:
Under instruction on the main page:
https://tutorial.cytron.io/2020/12/29/raspberry-pi-zero-usb-webcam/ (dead)

I have a pi zero 2 but I'm no programmer so I fear I cannot solve the issues.

kbingham, 2023-08-09:
Ok - looks like someone reported this here: https://github.com/showmewebcam/showmewebcam/issues/219

Projects can only live when people work on them ;-) Perhaps you could make a PR / proposal to update the documentation and have a local copy of the guide that the cytron has taken down?

NKT00, 2024-01-03:
The issue with that is that the sum total of the instructions are here: https://github.com/showmewebcam/showmewebcam#instructions

```
Instructions

    [Follow the Cytron tutorial](https://tutorial.cytron.io/2020/12/29/raspberry-pi-zero-usb-webcam/).
    Smile & Enjoy!
```
so it is impossible to even get started.

kbingham, 2024-01-03:
Looks like that was removed indeed. Wayback machine has a copy - so perhaps it should be copied to the wiki:

Here's the wayback URL:
https://web.archive.org/web/20221008011400/https://tutorial.cytron.io/2020/12/29/raspberry-pi-zero-usb-webcam/

NKT00, 2024-01-04:
Ok, sorted it. Pull request sent. I've added a link to the archive, and also spent an hour writing and testing the process on Windows 10 and a couple of Zeros, and added the details into the readme, as well as some trouble shooting advise for Windows.

closed 2024-01-13 by dgsiegel