showmewebcam/showmewebcam #223: start.elf: is not compatible

saner62-liu, 2023-10-11:
After getting the error message start.elf not compatible from the "sdcard-raspberrypi4-8e9f9ac.img.zip" v.1.91 file we tried to build a new file (see all the steps bellow) but it still resulted in the same error message. Therefore I wonder if someone knows how to get showmewebcam to work on a Raspberry Pi 4 model B?

git clone --recurse-submodules https://github.com/showmewebcam/showmewebcam.git
cd showmewebcam
./build-showmewebcam.sh raspberrypi4 1> ../output.txt 2> ../error.txt

Took a new 64 GB sdcard and used rpi-imager for showmewebcam/output/raspberrypi4/images/sdcard.img
In my Raspberry Pi 4 Model B - 4GB i get the following information in a boot loop:
bootloader: 8ba17717 2023/01/11
update-ts: 1696584140
(Some more information displayed that I can write down if necessary.)

Read config.txt
Read start.elf
Read fixup.dat
Firmware: 4b4aff21f72c5b9ba39d83c7b0f8fa910a6ef99b Dec 15 2020 18:29:44
start.elf is not compatible

connor3485, 2023-11-09:
I'm getting the same on my 4B. I'm thinking this project is dead and they're not supporting 4B, sadly.

dgsiegel, 2023-11-09:
> I'm getting the same on my 4B. I'm thinking this project is dead and they're not supporting 4B, sadly.

The Pi 4B is supported, but it seems that the new firmware requires some changes. Patches are always welcome.

connor3485, 2023-11-09:
The Pi needs a firmware update or the image does?

dgsiegel, 2023-11-09:
It's kinda a catch-22 situation. Most Pi's have a new firmware that requires a newer kernel with libcamera but remove support for the older V4L interface. Showmewebcam requires that. The libcamera port has progressed nicely, but we ran into performance issues as currently libcamera runs on the CPU.

connor3485, 2023-11-09:
Performance issues even with the 4B? Where do I find that version?

dgsiegel, 2023-11-09:
Didn't test the 4B specifically, as I don't have it available, but the work is here #197

connor3485, 2023-11-09:
I'd be happy to test on my 4B if someone can point me to where the image is or how to build it.

kbingham, 2023-11-09:
The pi4 should run fine with the sofware encoders we had in that libcamera development branch. There is a hardware encoder for the JPEG that we can use too on Pi0 and Pi4 - it 'just' needs someone to work on it.

On Pi5 - the only software stack available is libcamera, so we'll need to figure out how to update somepoint. libcamera has coexisted with legacy for the last couple of years to allow the software to start getting ported over - but from pi5 onwards - there is no 'legacy' support. So whatever future generations of Pi device are created - to support them we need to move away from the legacy camera stack.

Oceanique, 2024-02-09:
Hello, I replaced both of the files "start.elf" and "fixup.elf" by the most recent ones and all I get is no video output. On my laptop, it's not even detected by dmesg -w .... 

I'm tired, really, it's been three days, I come from this project which worked fine two years ago : https://github.com/geerlingguy/pi-webcam but now, even trying with old RPiOS like Buster or Bullseye, it doesn't work.