showmewebcam/showmewebcam #228: Camera module 3 support ?

ABelliqueux, 2024-01-10:
Is the  Camera module 3 (imx708) supported ?
If not, is it planned ?

kbingham, 2024-01-10:
showmewebcam presently uses the legacy camera stack from Raspberry Pi. The legacy camera stack can not operate the IMX708.

We have done work to update showmewebcam to the new libcamera stack, but the performances are low - as presently it uses a software encoder. There is a hardware encoder available on the Pi0-2, but the work we have done needs to be extended to make use of that.

I successfully use the IMX708 with the software encoder on an RPi0-2W - but it's not likely to work well at all on an RPi0.

So - 'planned' yes - if someone can dedicate the time to help support this, and there's even preliminary work already functional. It needs someone to work on it.

ABelliqueux, 2024-01-10:
I'm trying to use a RPI 4B, do you think I could use the software encoder on that ? 
If so, is there an artifact/img I can get to test it ?
By low performance, you mean low framerate I suppose ?

kbingham, 2024-01-10:
Yes, it should work on a Pi4B with the software encoder. The hardware encoder can also be developed on there once you get started. You'll have to self build at the moment I'm afraid.

ABelliqueux, 2024-01-11:
Ok, tried to build but got :

```
../output/raspberrypi4/build/libcamera-ab72e6641c56e876f91edee57a8969982a3ab9b7/include/libcamera/meson.build:84:25: ERROR: Index 2 out of bounds of array of size 2.
```

which corresponds to 
```python
# version.h
version = libcamera_version.split('.')
libcamera_version_config = configuration_data()
libcamera_version_config.set('LIBCAMERA_VERSION_MAJOR', version[0])
libcamera_version_config.set('LIBCAMERA_VERSION_MINOR', version[1])
libcamera_version_config.set('LIBCAMERA_VERSION_PATCH', version[2])
```

so my guess is that `libcamera_version` is that the currently used libcamera version is only 2 chars long (x.y) instead of three (x.y.z) ?

I'm not sure where  `libcamera_version` is sourced from though, so I can't check...

Any hint ?

dgsiegel, 2024-01-11:
> Any hint ?

What Buildroot version are you using? In #197 I didn't set any version for libcamera and it builds just fine.

ABelliqueux, 2024-01-11:
@dgsiegel : 
See my updated message above for more inos.

About buildroot version how do i find out ?

All I didi was : 

```
git clone --recurse-submodules https://github.com/showmewebcam/showmewebcam.git
git checkout buildroot-update
./build-showmewebcam.sh raspberrypi4
```

dgsiegel, 2024-01-11:
> About buildroot version how do I find out ?

Buildroot is a submodule in this repo, so you'd find out the current version via git. It might make sense to update to a newer version though, see my linked PR.

> 
> After some investigation, it looks like libcamera_version is sourced via the utils/gen-version.sh script, l.13:
> 
> ```version=$(git describe --abbrev=8 --match "v[0-9]*" 2>/dev/null)```
> 
> which when ran manually yields :
> 
> ```v1.91-10-g54cad4ee```
> 
> which does not conform to the pattern described in meson.build : e.g. 1.2.3+211-c94a24f4 .

Ah yeah, I remember running into this problem as well. `v1.91-10-g54cad4ee` is the showmewebcam version and is set via Buildroot. Libcamera thinks this is their version string too and breaks. I think this was fixed in more recent libcamera builds, and hence upgrading Buildroot should solve that.

ABelliqueux, 2024-01-11:
Ok, I'm getting further along the way. Here is what I did : 

```
git clone --recurse-submodules https://github.com/showmewebcam/showmewebcam.git
cd showmewebcam
git checkout buildroot-update
cd buildroot 
git checkout 9da87c
cd ..
./build-showmewebcam.sh raspberrypi4
```

I still encountered the index out of bound error and had to edit the `meson.build` file, l.84 : 

```
libcamera_version_config.set('LIBCAMERA_VERSION_PATCH', version[2])
```
to : 
```
libcamera_version_config.set('LIBCAMERA_VERSION_PATCH', '0')
```
but I am now getting multiple errorss when trying to build : 

Undeclared variable:
```
cap_proc.c:530:16: error: 'CAP_MODE_HYBRID' undeclared (first use in this function); did you mean 'CAP_MODE_NOPRIV'?
```
and conflicting types : 
```
cap_proc.c:850:5: error: conflicting types for 'cap_launcher_callback'; have 'int(struct cap_launch_s *, int (*)(void *))'
  850 | int cap_launcher_callback(cap_launch_t attr, int (callback_fn)(void *detail))
      |     ^~~~~~~~~~~~~~~~~~~~~
In file included from libcap.h:17,
                 from cap_proc.c:23:
/home/arthus/build/showmewebcam/output/raspberrypi4/host/include/sys/capability.h:188:13: note: previous declaration of 'cap_launcher_callback' with type 'void(struct cap_launch_s *, int (*)(void *))'
  188 | extern void cap_launcher_callback(cap_launch_t attr,
      |             ^~~~~~~~~~~~~~~~~~~~~
cap_proc.c:865:5: error: conflicting types for 'cap_launcher_setuid'; have 'int(struct cap_launch_s *, uid_t)' {aka 'int(struct cap_launch_s *, unsigned int)'}
  865 | int cap_launcher_setuid(cap_launch_t attr, uid_t uid)
      |     ^~~~~~~~~~~~~~~~~~~
/home/arthus/build/showmewebcam/output/raspberrypi4/host/include/sys/capability.h:190:13: note: previous declaration of 'cap_launcher_setuid' with type 'void(struct cap_launch_s *, uid_t)' {aka 'void(struct cap_launch_s *, unsigned int)'}
  190 | extern void cap_launcher_setuid(cap_launch_t attr, uid_t uid);
      |             ^~~~~~~~~~~~~~~~~~~
cap_proc.c:882:5: error: conflicting types for 'cap_launcher_setgroups'; have 'int(struct cap_launch_s *, gid_t,  int,  const gid_t *)' {aka 'int(struct cap_launch_s *, unsigned int,  int,  const unsigned int *)'}
  882 | int cap_launcher_setgroups(cap_launch_t attr, gid_t gid,
      |     ^~~~~~~~~~~~~~~~~~~~~~
/home/arthus/build/showmewebcam/output/raspberrypi4/host/include/sys/capability.h:191:13: note: previous declaration of 'cap_launcher_setgroups' with type 'void(struct cap_launch_s *, gid_t,  int,  const gid_t *)' {aka 'void(struct cap_launch_s *, unsigned int,  int,  const unsigned int *)'}
  191 | extern void cap_launcher_setgroups(cap_launch_t attr, gid_t gid,
      |             ^~~~~~~~~~~~~~~~~~~~~~
cap_proc.c:902:5: error: conflicting types for 'cap_launcher_set_mode'; have 'int(struct cap_launch_s *, cap_mode_t)' {aka 'int(struct cap_launch_s *, unsigned int)'}
  902 | int cap_launcher_set_mode(cap_launch_t attr, cap_mode_t flavor)
      |     ^~~~~~~~~~~~~~~~~~~~~
/home/arthus/build/showmewebcam/output/raspberrypi4/host/include/sys/capability.h:193:13: note: previous declaration of 'cap_launcher_set_mode' with type 'void(struct cap_launch_s *, cap_mode_t)' {aka 'void(struct cap_launch_s *, unsigned int)'}
  193 | extern void cap_launcher_set_mode(cap_launch_t attr, cap_mode_t flavor);
      |             ^~~~~~~~~~~~~~~~~~~~~
cap_proc.c:945:5: error: conflicting types for 'cap_launcher_set_chroot'; have 'int(struct cap_launch_s *, const char *)'
  945 | int cap_launcher_set_chroot(cap_launch_t attr, const char *chroot)
      |     ^~~~~~~~~~~~~~~~~~~~~~~
/home/arthus/build/showmewebcam/output/raspberrypi4/host/include/sys/capability.h:195:13: note: previous declaration of 'cap_launcher_set_chroot' with type 'void(struct cap_launch_s *, const char *)'
  195 | extern void cap_launcher_set_chroot(cap_launch_t attr, const char *chroot);
```

Do you have a specific hash commit for buildroot that I should use ?

dgsiegel, 2024-01-11:
> Do you have a specific hash commit for buildroot that I should use ?

In #197 I've used Buildroot 2023.02: https://github.com/showmewebcam/showmewebcam/pull/197/commits/d467a382e3283e0a0add7a484c0a428f02298bad

I guess you could use any newer stable Buildroot release.

ABelliqueux, 2024-01-11:
Yes, that's the one I'm currently trying to use (commit `8cca1e`) but I now have a new error coming up : 

```
extend.texi:2490: @itemx should not begin @table
make[3]: *** [Makefile:3385: doc/gcc.info] Error 1
make[3]: *** Waiting for unfinished jobs....
In file included from ../../gcc/spellcheck-tree.h:23,
                 from ../../gcc/c/c-decl.c:55:
In member function 'edit_distance_t best_match<GOAL_TYPE, CANDIDATE_TYPE>::get_cutoff(size_t) const [with GOAL_TYPE = tree_node*; CANDIDATE_TYPE = tree_node*]',
    inlined from 'best_match<GOAL_TYPE, CANDIDATE_TYPE>::candidate_t best_match<GOAL_TYPE, CANDIDATE_TYPE>::get_best_meaningful_candidate() const [with GOAL_TYPE = tree_node*; CANDIDATE_TYPE = tree_node*]' at ../../gcc/spellcheck.h:175:38,
    inlined from 'name_hint lookup_name_fuzzy(tree, lookup_name_fuzzy_kind, location_t)' at ../../gcc/c/c-decl.c:4405:48:
../../gcc/spellcheck.h:162:39: warning: 'bm.best_match<tree_node*, tree_node*>::m_best_candidate_len' may be used uninitialized [-Wmaybe-uninitialized]
  162 |     return ::get_edit_distance_cutoff (m_goal_len, candidate_len);
      |            ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
../../gcc/c/c-decl.c: In function 'name_hint lookup_name_fuzzy(tree, lookup_name_fuzzy_kind, location_t)':
../../gcc/c/c-decl.c:4311:26: note: 'bm.best_match<tree_node*, tree_node*>::m_best_candidate_len' was declared here
 4311 |   best_match<tree, tree> bm (name);
      |                          ^~
/bin/sh ../../gcc/../move-if-change tmp-automata.c insn-automata.c
echo timestamp > s-automata
/bin/sh ../../gcc/../move-if-change tmp-attrtab.c    insn-attrtab.c
/bin/sh ../../gcc/../move-if-change tmp-dfatab.c     insn-dfatab.c
/bin/sh ../../gcc/../move-if-change tmp-latencytab.c insn-latencytab.c
echo timestamp > s-attrtab
make[2]: *** [Makefile:4427: all-gcc] Error 2
make[1]: *** [package/pkg-generic.mk:293: /home/arthus/build/showmewebcam/output/raspberrypi4/build/host-gcc-initial-11.3.0/.stamp_built] Error 2
make: *** [Makefile:23: _all] Error 2
```
This one is quite obscure and I think that's where I'll stop trying. Thanks for your time and answers though!

kbingham, 2024-01-12:
That looks like errors building GCC ... which is buildroots responsibility. I would suspect it needs buildroot updating to a more recent version. If this is a bug exposed by buildroot. They would have already fixed it.

ABelliqueux, 2024-01-12:
Finally got it working after switching to buildroot 2023.08 : 

So here are the final steps to get it built : 

```
git clone --recurse-submodules https://github.com/showmewebcam/showmewebcam.git
cd showmewebcam
git checkout buildroot-update
cd buildroot 
git checkout 2023.08
cd ..
./build-showmewebcam.sh raspberrypi4
```

Testing the resulting image right now :) 

Thanks again to both of you.

Should I edit the issue's title to reflect the main discussion ?

kbingham, 2024-01-12:
Excellent! Keep us posted. I'll help where ever I can.

ABelliqueux, 2024-01-12:
It boots fine but doesn't seem to find the imx708, are there some steps I missed ?

In  `/boot/overlays`, there is no `imx708.dtbo` file and no module with that name either in `/lib/modules/5.10.92-v7l/kernel/drivers/media/i2c/`... How would I go about adding it ?

Here are some logs : 
```
$ cam -l
[0:12:01.248106680] [272]  INFO Camera camera_manager.cpp:298 libcamera v1.91+23-a5116c27-dirty (2024-01-12T15:51:32+01:00)
Available cameras:

$ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/media1

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media0
        /dev/media2

Cannot open device /dev/video0, exiting.

$ journalctl -u uvc-webcam.service 
Jul 28 08:26:33 piwebcam systemd[1]: Started Starts Raspberry Pi UVC webcam serv
Jul 28 08:26:33 piwebcam piwebcam[219]: One video device is missing, will not st
Jul 28 08:26:33 piwebcam systemd[1]: uvc-webcam.service: Main process exited, co
Jul 28 08:26:33 piwebcam systemd[1]: uvc-webcam.service: Failed with result 'exi
```

kbingham, 2024-01-12:
> In /boot/overlays, there is no imx708.dtbo file and no module with that name either in /lib/modules/5.10.92-v7l/kernel/drivers/media/i2c/... How would I go about adding it ?

A 5.10 kernel is likely far too old. It needs to be updated to the point that Raspberry Pi added the IMX708. The current raspberry pi kernel should either by rpi-6.1-y or rpi-6.6-y.

> [0:12:01.248106680] [272]  INFO Camera camera_manager.cpp:298 libcamera v1.91+23-a5116c27-dirty (2024-01-12T15:51:32+01:00)

And along with that - libcamera would likely need to be updated as well, but with the versioning bug - I can't tell what version of libcamera is being used there.

ABelliqueux, 2024-01-12:
Ok, so after looking at [225](https://github.com/showmewebcam/showmewebcam/issues/225) and https://github.com/showmewebcam/showmewebcam/issues/223, I modified  a few things : 

## rpi-firmware 

Replace files in `buildroot/package/rpi-firmware` with those : 

https://github.com/buildroot/buildroot/tree/master/package/rpi-firmware

## rpi-linux (kernel)

In `configs/config` : 
```
BR2_LINUX_KERNEL_CUSTOM_TARBALL_LOCATION="$(call github,raspberrypi,linux,a984fda6b2c24dbf1ca21924f99c8f9418f5765e)/linux-a984fda6b2c24dbf1ca21924f99c8f9418f5765e.tar.gz"
# Linux headers same as kernel
BR2_PACKAGE_HOST_LINUX_HEADERS_CUSTOM_6_1=y
```

## libcamera 

Replace files in `buildroot/package/libcamera` with those : 

https://github.com/buildroot/buildroot/tree/master/package/libcamera

## bootcode packaging

Add `BR2_PACKAGE_RPI_FIRMWARE_BOOTCODE_BIN=y` to `configs/raspberrypi4`.

## Rebuild

Not sure if the doc (https://github.com/showmewebcam/showmewebcam/wiki/Development-guide#sample-scenario-uprevving-uvc-gadget)  is up to date but running `BOARDNAME=raspberrypi4 make -C "output/$BOARDNAME" rpi-firmware-rebuild` from `showmewebcam` didn't work for me, so I had to do : 

```
rm -rf output/raspberrypi4
```

before building everything again. 

It's building right now :crossed_fingers: ...

ABelliqueux, 2024-01-13:
Well it built fine, but the rpi won't start with that image. All I'm getting is a blinking cursor after the rainbow screen. Not sure how I can find the culprit from here...

The img file is here fwiw : https://upload.disroot.org/r/HB_nSM95#nKXBwg4/J1jkQ+mvdbOYZeYfhM1c6JVnlFCCZHO9DhM=

The-God-coder, 2025-02-14:
Has any progress been made towards supporting the camera module V3 as of now?

The-God-coder, 2025-02-15:
I wanted to be able to use an embedded linux solution since Raspbian is a little slow for my specific needs. I guess my question would be has there been any progress in updating the kernel that this project uses to a more recent version?