showmewebcam/showmewebcam #229: Nothing is happening on my RPi 4 model B

Oceanique, 2024-02-09:
Hello,

I come from this project : https://github.com/geerlingguy/pi-webcam which doesn't work anymore (it was working fine in 2022 when I built it) and someone recommended the showmewebcam project.

But after a build that took ages that I put on my SD card, it wouldn't boot. I knew by reading another issue I had to replace the fixup.dat file and start.elf file by the most recent ones, but then, nothing on my monitor, it wasn't even detected by dmesg -w command when I connected my RPi to my laptop... I'm desperate, I only bought the HQ camera and the RPi to turn it into a HQ webcam for astronomy and taking pictures of wild life :'(

Thank you

dgsiegel, 2024-02-09:
Either take an already built image or help us port this project to the new Raspberry firmware/libcamera stack.

closed 2024-02-09 by dgsiegel

Oceanique, 2024-02-09:
I don't have enough experience in computer science to help badly... Could you tell me where the already built images are ? Thank you

dgsiegel, 2024-02-09:
It's described in the README, please have a read: https://github.com/showmewebcam/showmewebcam/releases

monkeymademe, 2024-02-09:
I do have an alt solution that runs on the bookworm but its not strictly a webcam but streaming over web interface. Maybe that floats boats?

https://github.com/monkeymademe/picamera2-WebUI-Lite