showmewebcam/showmewebcam #231: Request: USB to HDMI

EngiPCS, 2024-04-25:
Can a customized version of this build work to take a USB UVC input and convert it to a HDMI signal?

htruong, 2024-06-12:
Theoretically yes, I have not tried it, but it should work. However, I think you'd have a much better time with something that does it commercially on Amazon. I have something like this

https://www.amazon.com/hdmi-capture-usb/s?k=hdmi+capture+usb

and it works wonderfully.

closed 2024-06-12 by htruong

EngiPCS, 2024-06-13:
I've look at that, but that's the opposite of what I need.
That's HDMI to USB not the other way around, which I need.
The are only a few devices you can but that can accomplish this.
Like: [https://www.amazon.com/OBSBOT-UVC-HDMI-Adapter-Webcams/dp/B0C7433DSK](url)

But those prices are insane.

htruong, 2024-06-13:
Sorry I misunderstood you. It's outside of the project's scope, but I think that something like this might work, although it might require some tinkering.

https://forums.raspberrypi.com/viewtopic.php?p=1236142&sid=afcfd652d9b1d9a5802dc03583014ab8#p1236142

kbingham, 2024-06-14:
> I've look at that, but that's the opposite of what I need.
> That's HDMI to USB not the other way around, which I need.
> The are only a few devices you can but that can accomplish this.
> Like: [https://www.amazon.com/OBSBOT-UVC-HDMI-Adapter-Webcams/dp/B0C7433DSK](url)
> 
> But those prices are insane.

If you want to just display a camera on the screen without a gui you can use our libcamera 'cam' tool. That's outside of the scope of this project still, but if you want to try it "cam -c1 -C -D" is I think all you need.