showmewebcam/showmewebcam #243: Help needed: Support for full resolution still pictures

FilipDominec, 2025-05-21:
**TLDR**: 
I want to try extending showmewebcam capabilities for taking still pictures. 

**Motivation**:
Dear all, let me first thank you for this project, it works like a charm. We have set up the Pi HD camera as the main detector in a low-light spectroscopic setup and _showmewebcam_ allows us to tune the optics in real time. 

However, real measurement will require to manually set ISO and shutter, and take one full-resolution 12Mpx uncompressed picture. After doing a short research, I understood there is no support for still pictures in showmewebcam. This is in accordance with the good old unix principle do one thing well. 

Still I would like to ask you about adding this functionality, because

1. Fast-starting and reliable minimalistic operating system developed for this project is exactly what a still picture camera would need as well.
2. Having webcam *and* still picture capabilities in one device can be very useful (e.g. for real-time alignment of astronomic telescopes or other optical instruments).
3. This project is fairly popular and people like me apparently come here first when experimenting with Pi Zero & Pi Camera. Making a separate project for still pictures would not only take more work, but would bring much confusion into the community. There should be one obvious way to do the job, and I believe I am on the right place. 

**Proposed approach**
With my limited knowledge I imagine a program should be added in the /bin/ in the Pi Zero system, that would accept binary input through the virtual serial connection (few numbers describing the photo mode: shutter, ISO, scaling...) and would return a binary blob of the raw data taken from the camera. This blob can be relatively easily saved as RAW file or loaded into e.g. a Python script for further processing. 

The very simplest approach for me would be to write this in Python, importing the https://pypi.org/project/picamzero/ and ```pyserial``` modules. However, I don't think you would like to add entire Python in the system. 

So perhaps more efficient would it be to compile a little C binary using the same libraries as picamzero does. Having written some programs for desktop or Pi Pico, I have however no experience with development for Pi Zero, so I would have to get into grips with the ecosystem first.

**I will be thankful for** 

- your telling me if this feature is viable and welcome in this project,
- pointing me to some already existing solution (if I missed it),
- linking me with to people who already attempt[ed] this, 
- giving me your opinion what's the best approach to accomplish this. 

The Raspberry ecosystem is great, and this feature can be extremely useful for the community. Thanks a lot.

FilipDominec, 2025-05-22:
Currently I am recompiling entire firmware from scratch, believing I will find some way how to include any raspistill or any means to capture long-exposure single frame. Any hints how to do this are appreciated.

elcalzado, 2025-05-23:
You mention my post on the Raspberry Pi forum where I discussed an error I was getting while compiling. While this isn't much help I'll try to explain what happened for me. After many failed attempts of getting it to compile, I started changing versions to see if the error came from a version mismatch. I used Raspberry Pi OS Lite to test that certain versions worked together (I chose the latest) and then tried compiling those in buildroot. The error I was getting went away "magically" after I randomly tried to compile again. This isn't really helpful because it doesn't really teach me what went wrong or how to fix it. But, since it worked I made sure to save that sdcard.img file and flashed it to my Pi Zero 2 W. There, I ran into a new issue which was:

```
ERROR: rpicam-apps currently only supports the Raspberry Pi platforms.
Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps.
```

I realized rpicam-apps wasn't actually necessary for the application I'm trying to run so I left it at that and went back to trying to figure out how to get libcamera to detect my camera.

Anyway, I took notes when I was doing it and I'll leave below what I wrote down when I got it to compile properly:

```
raspberrypi/libcamera version: d83ff0a4ae4503bc56b7ed48cd142c3dd423ad3b
	- hash: 8492befa0f55a499269e3efdd6c03313917a37bf184bb57552ede6a6fc4d6d7d  libcamera-d83ff0a4ae4503bc56b7ed48cd142c3dd423ad3b-git4.tar.gz
raspberrypi/rpicam-apps version: 1.7.0
	- hash: 7374950690b9bec8b9eb3fa4af37c397e4082d6cca5a8160f6584dcdcea4f51b  rpicam-apps_1.7.0.orig.tar.xz

The controls::rpi error went away "magically" all I did was try building libcamera-apps 1.5.0, fail, then try libcamera-apps 1.7.0 and it worked.
```

If there's anything I might be able to help with let me know and I can try but I'm new to this stuff. If you have any info on how to get libcamera to detect the camera on buildroot please let me know. [Here](https://gitlab.freedesktop.org/camera/uvc-gadget/-/issues/24) i discuss what problems I'm currently facing with my project.